wpilibsuite/sysid #494: Network Tables will not connect. 2023.4.1 and 2023.4.2

JustJoshingYa, 2023-02-26:
Hi. Our team is trying to get feed forward values for our arm. We are now using a AMT103 Encoder (prr 512, cpr 2048) on the end of our motor shaft, which plugged into our spark max which is still brushed mode since we are still using it with our PG71 775 motor. Our gear ratio is 366.36:1, our We have verified everything electrically works via continuity and the rev hardware client.

[image]

However, when we try to run sysid with it, we find we can’t connect to network table with it. We’ve restarted our computer, changed our NT ports to default for NT3 and NT4, usb connected to robot, connected via radio, changed samples per average to 8, upgraded from wpilib 2023.4.1 to 2023.4.2, changed the ip, and even downloaded MVSC 2022 community edition to try to compile it ourselves.

When we deploy the sysid code, the sysid console says we deployed successfully. HOWEVER, when we look at the rio log, it said the project failed because the encoder time measurement window is too low, it must between 1 - 100. (I don't have screenshots at the moment) But there is no setting for encoder time measurement in generate for spark max brushed.

Other than that, we have no idea why network tables does not work. Sysid have worked for us in the past for getting pid values for our falcon 500 mk4 swerve drive running 2023.2.3. It also does show in sysid it connects when we deploy our own code, but not when we deploy sysid code with the same ip that we set for our own code.

calcmogul, 2023-02-26:
NetworkTables connections in SysId have been flaky for years, but it's been especially bad this year.

JustJoshingYa, 2023-02-26:
Why is the rio log saying the encoder time measurement is too low and failing the project deploy if I can't edit it?

calcmogul, 2023-02-26:
See if https://github.com/wpilibsuite/sysid/pull/495 fixes it.

JustJoshingYa, 2023-02-26:
I'll test it on Monday.

PeterJohnson, 2023-02-26:
Is robot code shown as running in the DS?  If the robot program is stopping because there's an error, NT isn't going to connect.

JustJoshingYa, 2023-02-26:
Robot code did not appear as running in the DS.

JustJoshingYa, 2023-02-26:
So does that mean since the encoder time measurement was prohibiting the sysid program from deploying fully, that's why NT would not connect?

JustJoshingYa, 2023-02-26:
> NetworkTables connections in SysId have been flaky for years, but it's been especially bad this year. We don't know why.
Ooh, I should also mention, when I hit default port after changing the port number while the sysid code was already deployed, it did attempt to connect and did show up as connected on sysid for half a second. 

> Is robot code shown as running in the DS? If the robot program is stopping because there's an error, NT isn't going to connect.

But I guess due to encoder time measurement being wrong by default, NT stopped which is why it only connected for 500 ms.

calcmogul, 2023-02-27:
You can use the Windows artifact for convenience.

JustJoshingYa, 2023-02-27:
Alright great, though I am curious, why did this error appear when we tried build the fork? 

Could not run phased build action using connection to Gradle distribution 'https://services.gradle.org/distributions/gradle-7.5.1-bin.zip'.
Script 'C:\Users\1810 Catatronics\Desktop\sysid\sysid-application\config\publish.gradle' line: 63
A problem occurred configuring project ':sysid-application'.
Exception thrown while executing model rule: publishing { ... } @ sysid-application\config\publish.gradle line 7, column 3
Cannot convert a null value to an object of type MavenArtifact.
The following types/formats are supported:
  - Instances of MavenArtifact.
  - Instances of AbstractArchiveTask, for example jar.
  - Instances of PublishArtifact
  - Instances of Provider
  - Maps containing a 'source' entry, for example [source: '/path/to/file', extension: 'zip'].
  - Anything that can be converted to a file, as per Project.file()

calcmogul, 2023-02-27:
I have no idea.

JustJoshingYa, 2023-02-27:
I found the issue, publish.gradle is set up for MacOS, not windows when we pull from the git repo. 
[image]

JustJoshingYa, 2023-02-27:
How would I change this to work on windows?

calcmogul, 2023-02-27:
I don't think that's the issue. Publish works in CI for Windows, macOS, and Linux. If it didn't, there would be no Maven artifacts to download for each OS.

calcmogul, 2023-02-27:
Besides, you can run `./gradlew run` to build and run the application locally without having to publish anything.

JustJoshingYa, 2023-02-27:
That's what I'm doing.

calcmogul, 2023-02-27:
Line 63 of publish.gradle says "artifact zip", so my guess is your machine has no way to zip and unzip things programmatically in a way Gradle understands. I don't know how one would go about fixing that.

JustJoshingYa, 2023-02-28:
We found that we didn't installed enough C++ packages from Visual Studio, so we did that. However, we now error here.
See file:///C:/Users/1810%20Catatronics/source/repos/sysid/sysid-library/build/tmp/compileLibsysidLinuxathenaReleaseStaticLibraryLibsysidCpp/output.txt for all output for compileLibsysidLinuxathenaReleaseStaticLibraryLibsysidCpp.
compiling SysIdSetup.cpp failed.
In file included from C:\Users\1810 Catatronics\source\repos\sysid\sysid-library\src\main\cpp\generation\SysIdSetup.cpp:5:
C:\Users\1810 Catatronics\source\repos\sysid\sysid-library\src\main\include/sysid/generation/SysIdSetup.h:18:10: fatal error: rev/CANSparkMax.h: No such file or directory
   18 | #include <rev/CANSparkMax.h>
      |          ^~~~~~~~~~~~~~~~~~~
compilation terminated.

compiling SysIdDrivetrainLogger.cpp successful.

compiling SysIdGeneralMechanismLogger.cpp successful.

compiling SysIdLogger.cpp failed.
C:\Users\1810 Catatronics\source\repos\sysid\sysid-library\src\main\cpp\logging\SysIdLogger.cpp:12:10: fatal error: ctre/Phoenix.h: No such file or directory
   12 | #include <ctre/Phoenix.h>
      |          ^~~~~~~~~~~~~~~~
compilation terminated.

Finished compileLibsysidLinuxathenaReleaseStaticLibraryLibsysidCpp, see full log file:///C:/Users/1810%20Catatronics/source/repos/sysid/sysid-library/build/tmp/compileLibsysidLinuxathenaReleaseStaticLibraryLibsysidCpp/output.txt.

JustJoshingYa, 2023-02-28:
Good news, both the version that I complied using my home computer and moved the complied working folder to the driver station and it ran. NT connected fine, and our arm did produce encoder  values. The windows artifact also worked. Thank you!

closed 2023-03-03 by PeterJohnson