wpilibsuite/sysid #498: Venom drivetrain crashes

wlmchen, 2023-03-17:
We are trying to characterize our drivetrain with SysID, but receive the following error when doing so. Our drivetrain consists of four venoms, and we are using the integrated encoders. We are using WPILib version 2023.4.2.

```
...
Setup Venom
Setup Venom
Setup encoder
Setup Built-in+Venom
Setup Built-in+Venom
Setup gyro
Setup None
Setting default
********** Robot program startup complete **********
Robot disabled
Collected: 0 data points.
Project failed: bad_function_call
[phoenix-diagnostics] Server shutdown cleanly. (dur:0)
[phoenix] Library shutdown cleanly
*** Error in `/home/lvuser/frcUserProgram': free(): invalid next size (fast): 0x0069ec88
```

When switching to general mechanism and testing a TalonFX, SysID works and shows `NT Connected.`

For reference, here is our [config file](https://pastebin.mozilla.org/SzNYXFr6)

calcmogul, 2023-03-17:
Support for Venom was removed for the most part, since a 2023 vendordep didn't exist during beta. If one exists now, we can copy that in and uncomment the support code.

wlmchen, 2023-03-17:
https://www.playingwithfusion.com/frc/playingwithfusion2023.json

PR #499 (2023-03-21): Add Venom support
Closes #498
closed 2023-03-21 by calcmogul