wpilibsuite/sysid #504: lots on NAN values in analysis

DAflamingFOX, 2023-03-28:
[image]
https://drive.google.com/drive/folders/15xvh2yAA24H4YBaGJ5t2FkWtN82asUdD?usp=share_link

calcmogul, 2023-03-28:
[config.json.txt](https://github.com/wpilibsuite/sysid/files/11094636/config.json.txt)
[sysid_data20230328-164517.json.txt](https://github.com/wpilibsuite/sysid/files/11094637/sysid_data20230328-164517.json.txt)

calcmogul, 2023-03-28:
The arm analysis says it used meters, for some reason. If I manually edit the JSON to use "Radians" instead, the analysis doesn't give NaN anymore.

calcmogul, 2023-09-28:
OBE by #518. The units are determined by the project JSON, and the project generator was removed.

closed 2023-09-28 by calcmogul